teeminus/NoTouchScreenFirmware #1: Is there a font generator here?

mbgroot, 2021-01-14:
[image]
On the TFT24 and klipper i see this

teeminus, 2021-01-14:
As far as I could see from the code, the character generator is implemented. I don't know if it is working, but after dumping the commands/data send to the display by klipper I only see the custom characters beeing programmed but not used.

Have you:
1) tried entering the menu by pressing the knob? How does that look?
2) tried restarting klipper host and firmware after flashing the LCD?

This is how my TFT35v3 looks with klipper:
[image]
[image]

mbgroot, 2021-01-14:
1. in the menu no font too.
2. yes, i'm restart.

teeminus, 2021-01-14:
Maybe the issue is caused by the fullscreen mode. I disabled that, please try again. If that does not work, I don't any more ideas why the firmware is not working for the TFT24.

mbgroot, 2021-01-14:
did not help :(

teeminus, 2021-01-14:
If have no idea what could cause the problems. The TFT24 uses a different (and slower) MSU than the TFT35. It could be timing related (the TFT24 has no fsmc high speed module). You could try increasing the delays in the klipper ST7920 driver.

teeminus, 2021-02-03:
The ST7920 driver has been reimplemented from scratch. The problem should be fixed now.

closed 2021-02-03 by teeminus