teeminus/NoTouchScreenFirmware #2: BIGTREE_TFT35_V3.0_E3

Sineos, 2021-02-03:
Commit: 9e32d731e2bc118e29bcdaf2b2336dc93e847faa
Display: BIGTREE_TFT35_V3.0_E3

Only showing "NoTouchFW V1.1" on a otherwise black screen. 

Thanks for keeping working on it!

teeminus, 2021-02-03:
Please try the binary from the revert_cleanup branch. Thats just the new parser added to the previous release (which seemed to partly work)

Sineos, 2021-02-03:
No luck unfortunately. Same behavior.

teeminus, 2021-02-03:
Place the following code snipped before `// Init slave SPI` in the main.cpp and recompile:
```
  const uint8_t pData[] = {0xf8, 0x30, 0x00, 0x00, 0x10, 0x00, 0x60, 0x00, 0xc0, 0xf8, 0x80, 0x00, 0xfa, 0x40, 0x10, 0x40, 0x10};
  for (int i = 0; i < sizeof(pData); ++i) {
    st7920Emulator.parseSerialData(pData[i]);
  }
  while(1);
```

The letter 'A' should appear twice on your screen.

Sineos, 2021-02-03:
Which branch you want me to try?

teeminus, 2021-02-03:
Please use the revert_cleanup branch. If the code snipped works there (at least shows the expected results) you can try it in the master branch :)

Sineos, 2021-02-03:
Works for both branches. I see a "NoTouchFW V1.1" heading followed by "AA" a bit below.

teeminus, 2021-02-03:
Thats great. At least the parser and the callback seems to work :) Can you move the snippet below the `SPI_Slave` call and check if that works as well?

BTW, how does the display section of you klipper config look like? For me it looks like this:
```
[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2
```

Sineos, 2021-02-03:
SKR1.4T
```
[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
```

But this was a perfect hint: I had a typo in there. Corrected now and what shall I say: IT WORKS 👍 

Looks good and also the fan animation works without flashing the display. Also switching to the menu comes up nicely.

Sorry for the noise and big kudos for your work.

teeminus, 2021-02-03:
Glad to hear that :pray: :relieved: Have fun with the display :relaxed:

P.S.: Is the config you posted correct? If not, please post the config here for others to find.

Sineos, 2021-02-03:
```
[board_pins]
aliases: 
	EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
	EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
	EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
	EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
```

This is my current SKR 1.4 Turbo configuration:
- Display works
- Encoder works
- Click works
- The only thing I noticed that sometimes after restarting Klipper, the display comes up garbled. Will require some service-restarts and/or firmware_restart

teeminus, 2021-02-03:
Thats possible, I only parsed the klipper display startup sequence once, I don't know what happens then the emulator gets scrambled data or if it's even the emulators fault. If that really bothers you you can create a new issue for that. I will look into that anyways.

Sineos, 2021-02-03:
No, doesn't bother me. Just wanted to give as feedback. 

Thanks for your commitment.

teeminus, 2021-02-03:
You are welcome :)

closed 2021-02-03 by teeminus

Sineos, 2021-02-03:
Side note:
It might be worthwhile to collect working configs somewhere. Maybe a section in the Github's Discussions, e.g. new Discussion item with the board name as title and just paste  [board_pins] and [display]

stevehooker, 2021-02-03:
> collect working configs somewhere

Yes please. And... Some beginner steps for flashing the firmware to the TFT screen. I've tried adding the precompiled BIGTREE_TFT24_V1.1.26.x.bin to an SD card, I expected that to be renamed to something like firmware.cur but alas no joy and I'm still on a black screen.

teeminus, 2021-02-04:
I created a wiki with both installation instructions and the configurations listed here. If you'd like to have any additions to the texts there, please let me know.