teeminus/NoTouchScreenFirmware #3: BTT_TFT24

MarsArtis, 2021-02-03:
used the  debug_TFT24 with STM linker file for STM32F105RCTx for TFT24.

Still getting black screen with **NOTOUCHFW V1.1** text

teeminus, 2021-02-03:
Try the revert_cleanup branch with the code snipped mentioned here: https://github.com/teeminus/NoTouchScreenFirmware/issues/2#issuecomment-772503695

Please report if the letter 'A' appears two times on the display.

MarsArtis, 2021-02-03:
built as per your instruction, but "AA" is not displayed, Still and only **NoTouchFW V1.1**

my .cfg is:
```
[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
```

teeminus, 2021-02-03:
Hmm, thats strange. You are using the revert_cleanup branch? You inserted the code between `Show title` and `Init slave SPI`, correct? You also copied the `while(1);` line?
You could try adding `GUI_Clear(RED);` after the for loop of the snippet and before the endless loop. Let's check if the controller reaches that line...

MarsArtis, 2021-02-03:
> You could try adding `GUI_Clear(RED);` after the for loop of the snippet and before the endless loop. Let's check if the controller reaches that line...

not clearly understood where to try and place `GUI_Clear(RED);`

this is how it looks the main.cpp
[image]

teeminus, 2021-02-03:
Like this:
```
  const uint8_t pData[] = {0xf8, 0x30, 0x00, 0x00, 0x10, 0x00, 0x60, 0x00, 0xc0, 0xf8, 0x80, 0x00, 0xfa, 0x40, 0x10, 0x40, 0x10};
  for (int i = 0; i < sizeof(pData); ++i) {
    st7920Emulator.parseSerialData(pData[i]);
  }
  GUI_Clear(RED);
  while(1);
```

Sorry for my communication skills, I am no native and have not spoken (or written) english for a long time. I didn't meant to be rude.

MarsArtis, 2021-02-03:
Neither I'm a native english speaker ;P
anyway I've added the 
> GUI_Clear(RED);

but no **AA** yet, Still black and **NoTouchFW V1.1 text**

teeminus, 2021-02-03:
Thats really strange... Try commenting out (or removing) the for loop that prints the title. If that does not work either, comment out everything that was added by the snipped except for the GUI_Clear command and the endless loop.

This question might be obvious, but are you copying the correct binary to SD card? Unfortunatelly the binary files in the binary folder are only updated when recompiling the firmware after the clean command has been issued. If you do not perform a clean build, the binary is only stored in the `.pio\build\...\` folder in the root of the repo.

MarsArtis, 2021-02-03:
> Thats really strange... Try commenting out (or removing) the for loop that prints the title. If that does not work either, comment out everything that was added by the snipped except for the GUI_Clear command and the endless loop.
> 
Have now removed the snippet but left only:
```
 GUI_Clear(RED);
  while(1);

```
and the screen get's all red with no text

> This question might be obvious, but are you copying the correct binary to SD card? Unfortunatelly the binary files in the binary folder are only updated when recompiling the firmware after the clean [...]
Well, after modifying the .cpp I've made a build (but no clean before!) and used the .bin output in the `.pio\build\...\ folder.
Am I not supposed to use this?

teeminus, 2021-02-03:
Yes, during debugging it is better to use the binary from the .pio\build\...\ folder to make sure you got the latest binary.

Could you please try the following snippet:
```
  // Show title
//  const uint8_t pTitle[] = {0x7F, 0x02, 0x04, 0x08, 0x7F, 0x38, 0x44, 0x44, 0x44, 0x38, 0x01, 0x01, 0x7F, 0x01, 0x01, 0x38, 0x44, 0x44, 0x44, 0x38, 0x3C, 0x40, 0x20, 0x7C, 0x00, 0x38, 0x44, 0x44, 0x44, 0x28, 0x7F, 0x04, 0x04, 0x78, 0x00, 0x7F, 0x09, 0x09, 0x09, 0x01, 0x3C, 0x60, 0x30, 0x60, 0x3C, 0x00, 0x00, 0x00, 0x00, 0x00, 0x1F, 0x20, 0x40, 0x20, 0x1F, 0x00, 0x42, 0x7F, 0x40, 0x00, 0x00, 0x60, 0x60, 0x00, 0x00, 0x00, 0x42, 0x7F, 0x40, 0x00};
//  for (uint16_t i = 0, x = (LCD_WIDTH - sizeof(pTitle) / 5 * 6) / 2; i < sizeof(pTitle); ++i, ++x) {
//    for (uint8_t y = 0; y < 8; ++y) {
//      if ((pTitle[i] & (1 << y)) > 0) {
//        GUI_FillRectColor(x, y, x + 1, y + 1, WHITE);
//      }
//    }
//    if ((i % 5) == 4) {
//      ++x;
//    }
//  }

  const uint8_t pData[] = {0xf8, 0x30, 0x00, 0x00, 0x10, 0x00, 0x60, 0x00, 0xc0, 0xf8, 0x80, 0x00, 0xfa, 0x40, 0x10, 0x40, 0x10};
  for (int i = 0; i < sizeof(pData); ++i) {
    st7920Emulator.parseSerialData(pData[i]);
  }
  while(1);

  // Init slave SPI
....
```

I would like to check if the parser and the callback function works.

MarsArtis, 2021-02-03:
> Could you please try the following snippet:

done.
Just blank black screen with no text at all.

teeminus, 2021-02-03:
I am currently out of ideas what to try next... Unfortuantelly I don't own a TFT24 to debug the issue myself.

Has the V1 version of this firmware worked for you (like the icons have been shown and the rest was scrambled)?

MarsArtis, 2021-02-03:
> I am currently out of ideas what to try next... Unfortuantelly I don't own a TFT24 to debug the issue myself.
> 
> Has the V1 version of this firmware worked for you (like the icons have been shown and the rest was scrambled)?

Yes, it was working quite well, no scramble text. Pretty clear. Just blinking

TheNitek, 2021-02-03:
I can confirm this behaviour: TFT24 with SKR1.3
V1: Icons work, rest is scrambled
V1.1: Screen doesn't change after NoTouch message

teeminus, 2021-02-03:
Try replacing the code of the main.cpp with the following:

```
#include "includes.h"

#include "myfatfs.h"
#include "ff.h"

#include "St7920Emulator.hpp"

void clearDisplay() {
  GUI_Clear(GREEN);
  while(1);
}
void drawByte(uint8_t x, uint8_t y, uint8_t d) {
  GUI_Clear(BLUE);
  while(1);
}

St7920Emulator st7920Emulator(clearDisplay, drawByte);

int main(void)
{
  // Set vector table offset
  SCB->VTOR = VECT_TAB_FLASH;

  // Get clock frequency
  RCC_ClocksTypeDef rccClocks;
  RCC_GetClocksFreq(&rccClocks);

  // Init NVIC priority group
  NVIC_PriorityGroupConfig(NVIC_PriorityGroup_2);

  // Init delay
  Delay_init(rccClocks.HCLK_Frequency);

  // Mount SD card
  if(mountSDCard())
  {
    // Check if firmware binary exists
    if (f_file_exists(FIRMWARE_NAME ".bin"))
    {
      // Check of old firmware binary exists
      if (f_file_exists(FIRMWARE_NAME ".CUR"))
      {
        // Delete old firmware binary
        f_unlink(FIRMWARE_NAME ".CUR");
      }

      // Rename current firmware binary
      f_rename(FIRMWARE_NAME ".bin", FIRMWARE_NAME ".CUR");
    }
  }

  // Init LCD
  LCD_Init();

  // Clear screen
  GUI_Clear(BLACK);

  const uint8_t pData[] = {0xf8, 0x30, 0x00, 0x00, 0x10, 0x00, 0x60, 0x00, 0xc0, 0xf8, 0x80, 0x00, 0xfa, 0x40, 0x10, 0x40, 0x10};
  for (int i = 0; i < sizeof(pData); ++i) {
    st7920Emulator.parseSerialData(pData[i]);
  }
  while(1);

  // Init slave SPI
  SPI_Slave();

  // Endless loop
  uint8_t data;
  while(true)
  {
    // Check if SPI data is available
    if (SPI_SlaveGetData(&data))
    {
      // Parse data
      st7920Emulator.parseSerialData(data);
    }
  }
}
```

Does the screen color change to anything else than black?

MarsArtis, 2021-02-04:
it fails to compile:

```
src/User/main.cpp: In function 'int main()':
src/User/main.cpp:59:21: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (int i = 0; i < sizeof(pData); ++i) {
                   ~~^~~~~~~~~
src/User/main.cpp:65:13: error: too few arguments to function 'void SPI_Slave(CIRCULAR_QUEUE*)'
   SPI_Slave();
             ^
In file included from src/User/includes.h:24:0,
                 from src/User/main.cpp:1:
src/User/Hal/stm32f10x/spi_slave.h:11:6: note: declared here
 void SPI_Slave(CIRCULAR_QUEUE *queue);
      ^~~~~~~~~
*** [.pio/build/BIGTREE_TFT24_V1_1/src/User/main.o] Error 1
Compiling .pio/build/BIGTREE_TFT24_V1_1/FrameworkHALDriver/Src/stm32f1xx_hal.o
============================================================================== [FAILED] Took 5.12 seconds ==============================================================================

Environment         Status    Duration
------------------  --------  ------------
BIGTREE_TFT24_V1_1  FAILED    00:00:05.123
======================================================================== 1 failed, 0 succeeded in 00:00:05.123 ========================================================================
The terminal process "pio 'run'" terminated with exit code: 1.
```

MarsArtis, 2021-02-04:
It fails if I leave make it build all the envs too

teeminus, 2021-02-04:
I updated the code in the comment: https://github.com/teeminus/NoTouchScreenFirmware/issues/3#issuecomment-772794752
Please try again.

MarsArtis, 2021-02-04:
compiled ok, flashed.
Dark screen with no text at all.

teeminus, 2021-02-04:
I think I found something. Apparently the .gitignore file prohibited, that the startup assembler code was committed to github. I added that locally for my TFT35 and now it stopped working as well. So maybe the wrong startup code is used while linking the binary.

MarsArtis, 2021-02-04:
Ok i'll look forward for your further updates. thx

teeminus, 2021-02-04:
So, I got it back working for my TFT35. I updated the [revert_cleanup](https://github.com/teeminus/NoTouchScreenFirmware/tree/revert_cleanup) branch with my changes. Please check out that branch and compile the firmware for your display.

MarsArtis, 2021-02-04:
at present, once flashed I get black screen with text:
**NoTouchFW V1.1** 
and two **BB**

teeminus, 2021-02-04:
Thats expected behaviour :) I changed that from AA to BB to see if my screen updates. The next two steps would be:
1. Remove the `// Show test data` paragraph and re-test that
2. Use the binary from the [add_startup_code](https://github.com/teeminus/NoTouchScreenFirmware/tree/add_startup_code) brach and test it together with Klipper. This branch is based on the current master and adds the startup code as well as moved the emulator handle to the main function. The binaries in that branch are precompiled and ready to use.

If you feel lucky enough you can skip step 1 and directly move to step 2 :D

MarsArtis, 2021-02-04:
> 2. Use the binary from the [add_startup_code](https://github.com/teeminus/NoTouchScreenFirmware/tree/add_startup_code) brach and test it together with Klipper. This branch is based on the current [...]
Tryed and got : **Illegal Flash APP!** on the TFT screen at the end of flashing

teeminus, 2021-02-04:
Hmm, thats strange. Try removing the SD card and see if the screen boots. If not, try recompiling the firmware yourself.

MarsArtis, 2021-02-04:
tried compiling commenting out the 
`// Show test data`
flashed and now it only shows **NoTouchFW V1.1**

Also tried to run klipper and restart the SKR board.

BTW the .bin downloadable from **add_startup_code_branch** is 109kb in size while the compiled one only 19Kb

MarsArtis, 2021-02-04:
sorry I want to add that NoTouchFW V1.1 remains on top of the screen

teeminus, 2021-02-04:
Which branch did you test?

teeminus, 2021-02-04:
> BTW the .bin downloadable from **add_startup_code_branch** is 109kb in size while the compiled one only 19Kb
I checked that but for me it's only 18k. Even recompiled it but no changes in size.

MarsArtis, 2021-02-04:
**revert_cleanup**,
built as per you instructions

teeminus, 2021-02-04:
Thats really nice :) :) :) Glad to hear that :)

I updated the add_startup_code_branch branch. It now shows an info text until it gets initialized by Klipper or Marlin. Could you please test that for me as well?

And could you please post your mainboard type and klipper config as well, please? I would like to add that to the wiki.

MarsArtis, 2021-02-04:
Tested the add_startup_code_branch branch 1.2:
flashed and got **NoTouchFW V1.2** followed by larger **ST920Emultor ready** text.

Restarted Klipper and LCD is displaying printer infos with tiny NoTouchFW V1.2 on the top of the screen.

It's connected to a SKR 1.4 Turbo and this is my config:

```
[display]
lcd_type: st7920 
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2

[output_pin beeper]
pin: EXP1_1

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
```

teeminus, 2021-02-04:
That good :)
I will then close this ticket and merge the changes to the master branch, okay?

Thanks for your Klipper config.

closed 2021-02-04 by teeminus

stevehooker, 2021-02-05:
Did you update the pre complied bins?
Still stuck with blank screen.

TheNitek, 2021-02-05:
V1.2 works perfectly for me

stevehooker, 2021-02-05:
This [BIGTREE_TFT24_V1_1](https://github.com/teeminus/NoTouchScreenFirmware/tree/main/binaries/BIGTREE_TFT24_V1_1)?

stevehooker, 2021-02-05:
Still nothing. Using the V1.2 of the BIGTREE_TFT24_V1_1 bin. Blank screen. I've tried a new power supply, new SD card, toggling power, hitting the button on the screen... 
Here's what I've done:
Added the binary to an SD card, popped it into the screen, toggled power, front button, restarted Klipper through Cura. I have the SKR1.4 and TFT24 pins in the config... All seems correct.

All was good with Marlin.  I didn't bother with the touch interface with Marlin as I kept getting errors with "finding the printer." I made do with the older Marlin interface and all was well. I don't know if that has anything to do with my problems.

I did get a spurious screen with the two options to chose between the old Marlin and the new touch screen interface, but just the once while toggling the front button, stuffing in the SD card... I found that odd, but unrepeatable. Again, I don't know if that has anything to do with my woes.

Here's the back of my screen, if there's something dumb...
[image]

Any and all clues, hints, tips, suggestions, gratefully received : -)

teeminus, 2021-02-05:
This screen looks like it's a TFT35_E3_V3 (at least from the silk screen printing). Please try the binary from the binaries/BIGTREE_TFT35_E3_V3_0 folder: https://github.com/teeminus/NoTouchScreenFirmware/blob/main/binaries/BIGTREE_TFT35_E3_V3_0/BIGTREE_TFT35_V3.0_E3.26.x.bin

stevehooker, 2021-02-05:
Doh!

Thank you. Yes, I get the Little text and the large ST7920Emulator ready... Just switch the config pins and I think I'll be OK.

Thanks again... Several printers here, with both TFT35s and TFT24s... My bad for mixing them up.