teeminus/NoTouchScreenFirmware #6: Feature Request: Brightness for supported models

RomRider, 2021-02-10:
Amazing work with this firmware, it just works perfectly :)

I'm trying to add brightness support through PWM, but it seems that the screen just blinks instead reducing the brightness. I'm probably doing something wrong as I'm not super familiar with the code.

I'm wondering if this would be something you could add and let the user define in some `.h` file for supported models?

Thanks a lot for your great work again!

teeminus, 2021-02-10:
If I would add that feature you would have to recompile it yourself as this wouldn't be a dynamic config option.

BTW.: Which display/mainboard combo are you using? If this combo is not listed [in the wiki](https://github.com/teeminus/NoTouchScreenFirmware/wiki/Klipper-configurations) please post the config for me to add to the wiki.

RomRider, 2021-02-10:
I'm fine with recompiling, I've done so already while experimenting with adding the feature back. It just doesn't dim, but blinks instead 😊

I'm using the TFT35 v3.

RomRider, 2021-02-10:
Thanks for your answer.

I've made it compile using:
* https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Hal/stm32f2_f4xx/timer_pwm.c
* https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Hal/stm32f2_f4xx/timer_pwm.h
(not only that but you get the idea)

But the result is that it blinks. What is it with reading the encoder?

teeminus, 2021-02-10:
If you just want to dim your display to a constant brightness you don't need the encoder. But if you want to dim your display when there has been no user interaction, then you need the encoder.

RomRider, 2021-02-10:
Yeah, I just want to reduce the brightness and always keep it this way, no brightness increase on user interaction etc... like on the initial firmware. The default 100% is very bright this is why I've tried experimenting with reducing it.

teeminus, 2021-02-10:
Implementing that is a piece of a cake :wink:

I created a feature branch for that for you to test: https://github.com/teeminus/NoTouchScreenFirmware/tree/add_static_dim_support
For your TFT35v3 you need the change the `LCD_LED_PWM_BRIGHTNESS` define [here](https://github.com/teeminus/NoTouchScreenFirmware/blob/add_static_dim_support/src/User/Variants/pin_TFT35_V3_0.h#L29) and recompile the firmware for that display.

RomRider, 2021-02-10:
Works perfectly, thanks a lot!
I did the same thing as you but I f***ed up with... `rccClocks` 🤦

closed 2021-02-10 by teeminus