teeminus/NoTouchScreenFirmware #8: Flip display vertically

jangrewe, 2021-02-10:
Hi, i tried `MKSTFT28.bin` on my Artillery Sidewinder X1 with a MKS TFT28 3.2, and it flashed without problems and now displays the "ST7920Emulator ready" screen, but i guess due to how the screen is built into the printer case, all text is upside down.
Can you maybe create a version that flips the output vertically?

teeminus, 2021-02-10:
Can you please show me a picture of your printer screen to clarify your situation? Is the screen rotated by 180° or is the screen flipped?

jangrewe, 2021-02-10:
Sorry, yes, it's flipped vertically and horizontally, or:  rotated by 180° 😆

[image]

teeminus, 2021-02-10:
Are you okay with compiling the firmware yourself? Then I could give you an **untested** solution for you problem.

jangrewe, 2021-02-10:
Sure, just put it in a branch and i'll give it a shot! 👍

teeminus, 2021-02-10:
Check out this branch: https://github.com/teeminus/NoTouchScreenFirmware/tree/add_rotate180_support
Uncomment the define on this line https://github.com/teeminus/NoTouchScreenFirmware/blob/add_rotate180_support/src/User/main.cpp#L13 and recompile for your display. I gave the fix a short test and it looked fine.

jangrewe, 2021-02-10:
Thanks, that worked perfectly! I still need to test the actual functionality, but at least i don't need to break my neck now 😁

closed 2021-02-10 by jangrewe

jangrewe, 2021-02-10:
Unfortunately it seems as if the MKS TFT28 is unusable with your firmware, as it only has a UART, not an SPI connection.
Maybe i'm completely off, but i consulted with a friend because i couldn't figure out the pins for the `[display]` section, and he said that even rewiring it to the MKS Gen L's EXP-1 header (instead of AUX-1) probably won't help much, because it doesn't speak SPI.
Am i missing something, or is it really completely hopeless?

Here are the datasheets:
Display: https://www.roboter-bausatz.de/media/pdf/04/6c/d6/MKS-TFT28-32-datasheet.pdf (page 10)
Board: http://sl3d.ch/wp-content/uploads/2019/11/MKS-Gen-L.pdf (page 8)

PS: sorry for self-hijacking this issue, i'm happy to open a new one if you prefer.

teeminus, 2021-02-11:
I would have prefered a new ticket for that but I am fine with discussing the issue here as well 😁 

TBH. I don't know any MKS screens or mainboards. What's made me curious is that the firmware in the BTT sources is named `MKS_TFT28_V1_0` but I could not find any information about that particular revision online. However there is a note in the BTT sources on how to use the screen in Marlin mode: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/923c5e01d41f08a0ea6945b2291ab5665bbf7b5c/TFT/src/User/Variants/pin_MKS_TFT32_V1_4.h#L82

It seems like that you only need the PB3 (SCK) and PB5 (MOSI) pins of the TFT as both the MISO and CS are disabled for the SPI and used by other parts of the TFT.

It should be obvious but just to be clear here: I do not advise you to solder on your TFT or mainboard. If you are brave enough to do that it's at your own risk.

However if you willing to go for it I am very interested in the results. And if it works I would appreciate if you share both your experiences (and maybe build steps) and your config with us 😄