teeminus/NoTouchScreenFirmware #13: ST7920 Emulator Ready

wildjokerss, 2021-02-12:
Hi there,

I'm facing the issue from the other case and although I've tried a few things none seem to get me out of the "ST7920 Emulator Ready". I got a SKR 1.4 non turbo with the TFT35 E3 V3 (think this is the correct name), setup seems pretty similar to yours / listed as supported or compatible but I'm having a hard time

```
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>
    # Pins EXP2_1, EXP2_6, EXP2_2 are also MISO, MOSI, SCK of bus "ssp0"
```
```
[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
#kill_pin: ^!EXP2_8

[output_pin beeper]
pin: EXP1_1

```

This is pretty much what I have in regards to making the screen work, do you see anything unusual or do you know how to get it going? I've made power cycles and firmware restarts etc but none seem to do the trick :(

I'd like to thank you in advance for doing this for the community, very appreciated!

teeminus, 2021-02-12:
Which mainboard are you using?

Have you ever seen the screen working in Marlin mode (like with the stock BTT firmware and Marlin)?

Have you tried different display configs? Like this one for the SKR 1.4T:
```
[display]
lcd_type: st7920 
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
```

wildjokerss, 2021-02-12:
what do you mean by mainboard firmware? Klipper? or are you talking about the one I flashed to the screen that you developed? that should be 1.2
Yeah sure, the screen always worked in TFT and Marlin mode with klipper, changed this to klipper recently.

I haven't tried it so far but now I have and IT FKING WOOOOORKS 

Dude thanks a ton for that!

closed 2021-02-12 by wildjokerss

teeminus, 2021-02-12:
For the logs: I wanted to know if you are running Klipper or Marlin 😉 

Glad it works, have fun 😄

JoeK1973, 2021-11-20:
Hi - I'm in the same position as @wildjokerss  - the TFT35 E3 3.0 never goes past ST7920Emulator ready message. I've got a SKR Mini E3 v1.2 and I can confirm it's worked fine  previously with Marlin 2.0.x and the standard TFT35 BTT firmware. Any hints? I'm really stuck

hapklaar, 2021-11-20:
> Hi - I'm in the same position as @wildjokerss - the TFT35 E3 3.0 never goes past ST7920Emulator ready message. I've got a SKR Mini E3 v1.2 and I can confirm it's worked fine previously with Marlin 2.0.x and the standard TFT35 BTT firmware. Any hints? I'm really stuck

Have you configured klipper to use the screen? I use the following lines in klipper.cfg for a SKR 1.3

```
[display]
lcd_type: emulated_st7920
spi_software_miso_pin: EXP2_1
spi_software_mosi_pin: EXP1_3
spi_software_sclk_pin: EXP1_5
spi_speed: 1000000
en_pin: EXP1_4
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
```