teeminus/NoTouchScreenFirmware #14: Dimming

Sineos, 2021-02-13:
Thanks for keeping it up. I can confirm that https://github.com/teeminus/NoTouchScreenFirmware/commit/37e3f05072bee8606ad9ce6f2cb3f8091a147846 works for my setup.

Is it possible to control dimming from within Klipper?

teeminus, 2021-02-13:
I'm sorry but I don't think this will works. Klipper assumes that the display is a real ST7920 which does not have that feature. I could add this feature to the firmware (there is an unused bit in the serial data preamble which could be used for that) but Klipper would need to implement the counter part for that. And I would have to investigate what a real ST7920 display would do with this invalid bit beeing set.

I think the stock BTT firmware dims the screen based on the user input (like rotating or clicking the knob). This feature could be readded if that would help.

Sineos, 2021-02-13:
Thanks for the explanation. Was expecting this.

My idea was to set the display to dimmed after a certain time of inactivity. Sort of "burn in prevention". Doing it manually will fail because of my laziness :wink:

teeminus, 2021-02-13:
I could implement a feature that turns the display backlight off after X seconds of inactivity if that would help 🤷 Rotating the know would turn the backlight back on of course 😀

Sineos, 2021-02-13:
This sounds great. Maybe not directly off but very low that it could be "barely" read (if possible) and back to full brightness on encoder or button.

teeminus, 2021-02-13:
Okay, give me a few days and I'll see what I can do 😄 😉

teeminus, 2021-02-14:
Check out this branch: https://github.com/teeminus/NoTouchScreenFirmware/tree/add_idle_dimmer
You need to edit the features.h and enable both PWM backlight dimming and display idle off feature there. After recompiling you are good to go. Pleaes let me know if both the brightness and the timeout is okay. As you are the first one to request this feature you are free to define those parameters 😄 😉 

This feature could be interesting for @RomRider as well 😄

RomRider, 2021-02-14:
Thanks, I'll have a look 😊 I wasn't missing that feature, but it's great it's there now! You rock!

Sineos, 2021-02-15:
Works 👍 
I'm using `LCD_LED_PWM_OFF_BRIGHTNESS 2`. Many thanks

teeminus, 2021-02-15:
Nice. It's in the master branch now 😄

closed 2021-02-15 by teeminus