teeminus/NoTouchScreenFirmware #15: BTT LCD 24 & SKR 1.3 Not working past Ready msg.

CraXyOW3, 2021-02-13:
I followed the previous thread on Klipper where you guided on how to make BTT's firmware to function in Marlin mode.
I compiled that one and everything worked, except some wierd menu artifacts and some other small stuff.

However, wanted to confirm the NoTouch working instead.
So i have tried both downlaoded the precompiled binary and i have also tried to compile myself.
In both cases the LCD is stuck at NoTouchFw V1.2 with ST7920Emulator ready.

Running the same pin config in Klipper as i did with the "original" firmware, reflashed it and that one works. But i want the NoTouch to function instead.

This is what i have in Klipper:
`[board_pins]
aliases:
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC

[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2`

Any tips?

teeminus, 2021-02-14:
Have you restarted klipper and the klipper mcu firmware after flashing the display?

There is a missing `>` at the end of `EXP2_10`. I don't think this will change anything but you never know...

From my expirience configs can a bit trick. Even though your config has worked with the stock BTT firmware, you could try this one as well: https://github.com/teeminus/NoTouchScreenFirmware/wiki/Klipper-configurations#tft35v3-and-btt-skr-v13

teeminus, 2021-02-16:
Are you capable of compiling the firmware from a dev branch yourself? Then you can clone into the following branch and test if that fixes your problem: https://github.com/teeminus/NoTouchScreenFirmware/tree/readd_cs_pin

There will be a new firmware release in a couple of days where this feature will most likely be included and that contains precompiled binaries as well.

CraXyOW3, 2021-02-18:
I finally got around to check it 😁 sorry for the wait.
I compiled the readd branch and now I got some artifacts too.
[image]

teeminus, 2021-02-18:
These artifacts are the SPI activity indicators. On the left side a pixel is drawn for each received byte, on the left side a line of 10 pixels is drawn for each activation of the SPI bus (when the CS pin is pulled high). When you press the knob and rotate is, you should see the left side update.

Speaking of the knob. Pressing the know for 5 seconds will reset the SPI interface. This is indicated by turning the screen backlight off and on again. Afterwards you can try the `firmware_restart` command from klipper.

CraXyOW3, 2021-02-19:
NOW that was some really weird stuff, Yesterday i rebooted everything, PI complete reboot, restart of both firmware and firmware_restart.
Power cycle and everything, even let it be dead for 30 minutes and then tried again.
Now when i came home i thought of testing the button press for 5 sec as you mentioned and now the screen works! O_o

Now when i can see the actual communication working with that bar above (artifacts :P ) i would wish to have them there or atleast a variable option when compiling it so one can choose to have it. Extremely useful even for seing if the mainboard is communicating.
Also, the 5sec button works.

So i am now successfully running SKR1.3 with BTT 2.4, NICE! Awesome Job.

Here is my current settings.

```
[board_pins]
aliases:
    EXP1_1=P1.30, EXP1_3=P1.18, EXP1_5=P1.20, EXP1_7=P1.22, EXP1_9=<GND>,
    EXP1_2=P0.28, EXP1_4=P1.19, EXP1_6=P1.21, EXP1_8=P1.23, EXP1_10=<5V>,
    EXP2_1=P0.17, EXP2_3=P3.26, EXP2_5=P3.25, EXP2_7=P1.31, EXP2_9=<GND>,
    EXP2_2=P0.15, EXP2_4=P0.16, EXP2_6=P0.18, EXP2_8=<RST>, EXP2_10=<NC>

[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2
```

These are the same settings as the "TFT24v1.1 and BTT SKR 1.4 Turbo" config. Same pins.

THANKS!

teeminus, 2021-02-19:
> NOW that was some really weird stuff, Yesterday i rebooted everything, PI complete reboot, restart of both firmware and firmware_restart.
> Power cycle and everything, even let it be dead for 30 [...]

Yeah, for some TFT/mainboard combinations the SPI of the display is not working properly on the first try. The root cause for that is pretty deep down the rabbit hole but I am working on that. There will be a new release (hopefully soon) that will make the screen work on first try - without the manual 5-second-knob-reboot trick.

> Now when i can see the actual communication working with that bar above (artifacts :P ) i would wish to have them there or atleast a variable option when compiling it so one can choose to have it. Extremely useful even for seing if the mainboard is communicating.

I added this feature for that exact reason. Its easier to figure out why a screen if not working when you know what works and where it does not work (anymore). As you may have noticed, there i a new horizontal line below the firmware name in the top. That line is drawn in two parts and it's only there for debug purposes 😅 

> Here is my current settings.

Thanks for that 😄 

> THANKS!

You are welcome 😉

closed 2021-02-19 by teeminus

teeminus, 2021-03-02:
There has been an update to both Klipper and this firmware that should fix your issue. Please checkout the latest release and read the release notes carefully 😄