teeminus/NoTouchScreenFirmware #16: BIGTREE_TFT35_V3.0.26.x.bin gives error "Illegal FLASH APP!"

ohjay93, 2021-02-13:
I copied the file BIGTREE_TFT35_V3.0.26.x.bin to a <2GB SD-card, and put it in the LCD card reader.
When turning on the printer it lists the file name and size, shows a percentage progress, but after reaching 100% only states
"Illegal FLASH APP!"

-Am I missing something here?

bakaufman, 2021-02-13:
flash once with .bin then remove the sd-card? I only get that on the second boot

teeminus, 2021-02-14:
I honestly don't know what causes this error. My guess would have been that the file has been corrupted while downloading but it could be the sd card, the firmware or the bootloader as well. I don't know when exactly this message is shown by the bootloader. It could be prior flashing or when booting the newly flashed app. Maybe there are some artifacts left on the sd card or the bootloaders finds something it interprets as a firmware file 🤷 

If removing the SD card after flashing does not fix that issue you could have redownloaded the binary or (as you did) compile it yourself.

Can this issue be closed now?

ohjay93, 2021-02-14:
It was super strange!
I tried re-downloading, different sd-cards, reformatting the cards and rebooting several times after the flashing failed.
My guess was that perhaps something had gone wrong when compiling the binaries, or perhaps that the bootloader refused them somehow.
Anyways I am really happy that it worked out in the end, so I am closing the issue now :-)

ohjay93, 2021-02-14:
Found a workaround: Compile the binaries myself.

closed 2021-02-14 by ohjay93