teeminus/NoTouchScreenFirmware #17: Initialize the PS_ON-Pin to HIGH

emtrax-ltd, 2021-02-14:
Can you initialize the PS_ON pin to HIGH on boot so that a connected BTT relay 1.2 does not automatically shuts down after 15 seconds? I know there is no point in keeping the relay connected this way, but it's the original wiring.

BIGTREETECH-TouchScreenFirmware/TFT/src/User/API/extend.c

```
// Power Supply Control pins Initialization
void PS_ON_Init(void)
{
  GPIO_InitSet(PS_ON_PIN, MGPIO_MODE_OUT_PP, 0);
  GPIO_SetLevel(PS_ON_PIN, infoSettings.ps_active_high);
}
```

teeminus, 2021-02-15:
I will have a look into that. Which TFT, mainboard and mainboard firmware (Marlin/Klipper) are you using?

emtrax-ltd, 2021-02-15:
- tft35 e3 v3.0
- skr mini e3 v1.2 ->since 2 days with klipper 0.9.x (based on the latest mainsailos-img with all upgrades afterwards, everything on a pi zero w), before that it was marlin 2.0.7

teeminus, 2021-02-15:
Can you please post your klipper display config here so that I can add that to the wiki?

I am currently working on an other issue but as soon as I finish that I will add the PS_ON feature. I think that will be done in a couple of days.

emtrax-ltd, 2021-02-15:
ps: you can check the "pin-high" with a voltage meter at ps-on-connector between ground and signal (should be finally above 3 V).

emtrax-ltd, 2021-02-15:
> Can you please post your klipper display config here so that I can add that to the wiki?
> 
> I am currently working on an other issue but as soon as I finish that I will add the PS_ON feature. I think that will be done in a couple of days.

```
[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2

[output_pin beeper]
pin: EXP1_1
```

currently, the py hack is running in klippers st7920.py - does not work reliably and flickers frequently

teeminus, 2021-02-15:
Thanks for the config.

I am curious about the hack, what does it do?

emtrax-ltd, 2021-02-15:
> Thanks for the config.
> 
> I am curious about the hack, what does it do?

it's "well known" hack from https://github.com/KevinOConnor/klipper/issues/2231

```
    def write_text(self, x, y, data):
        if x + len(data) > 16:
            data = data[:16 - min(x, 16)]
        pos = [0, 32, 16, 48][y] + x
        #self.text_framebuffer[pos:pos+len(data)] = data
        for c in data:
            self.write_graphics(x, y, font8x14.VGA_FONT[ord(c)])

```

teeminus, 2021-02-15:
Ah, okay. That one 😄

teeminus, 2021-02-20:
I havn't forgotten your request, I will take care about that on monday.

teeminus, 2021-02-21:
Well, it's not monday yet but I created a feature branch for you to test: https://github.com/teeminus/NoTouchScreenFirmware/tree/add_ps_on

You will have to compile the firmware yourself as I don't like pushing binary files to github, so I only update the binaries when creating a new release.

emtrax-ltd, 2021-02-21:
great job... it works!

lost 1h in testing... after ripping all apart too ... until i noticed, that i'm in the wrong branch for compiling *lol*

BOARD: BTT SKR MINI E3 V1.2
TFT: BTT TFT35 E3 V3
KLIPPER: encoder and beeper are functional
[display]
lcd_type: st7920
cs_pin: PB8
sclk_pin: PB9
sid_pin: PB7
encoder_pins: ^PA10, ^PA9
click_pin: ^!PB6

[output_pin beeper]
pin: PB5

teeminus, 2021-02-22:
Thanks for the feedback and the config. I would love to use the config for the [wiki](https://github.com/teeminus/NoTouchScreenFirmware/wiki/Klipper-configurations) but all screens there use the `board_pins` config section to define pin alias names. Could you add this section as well, please 😄

emtrax-ltd, 2021-02-22:
> Thanks for the feedback and the config.
> 
> EDIT: I just noticed that you posted the alias pins above. Are they still correct?

both cfg versions are effectively the same. however, i have noticed a some instabilities - the display does not always activate reliably. I'll investigate this in more detail over the next days.

otherwise the function for ps-on is 100% ok and so this issue is solved.

I would contact you when I got more information about the problem about the cfg and activation problem.

teeminus, 2021-02-23:
- That's a known problem. I have been working on that for the last week. There will be an update to both Klipper and this firmware that will hopefully fix the issue.

closed 2021-02-23 by teeminus