teeminus/NoTouchScreenFirmware #19: BIGTREETECH TFT35 V2.0

tgiphil, 2021-02-20:
Is v2.0 of BIGTREETECH TFT35 supported?

teeminus, 2021-02-20:
Unfortunatelly not. If it would be supported by BTT, I would have added it to this FW.

In the BTT sources there is a files called `pin_TFT35_V2_0.h`. This is the file where the capabilities of that screen is defined. There is a define named `ST7920_SPI` which has been commented out. Thats the reason why your screen is not supported by this FW. Unfortunatelly I cannot tell you if this firmware could support your screen if this define would not be commented out as I cannot test it. You could try compiling the original BTT firmware with that define not commented out yourself. But I cannot make any promises if this will work or even compile. And of course you do that on your own risk.
If the screen works with that firmware I can add support for it to this FW.

closed 2021-02-20 by teeminus

tgiphil, 2021-02-20:
@teeminus Thanks for the quick reply!