teeminus/NoTouchScreenFirmware #20: BTT SKR Mini 2.0 and TFT35-e3 v3.0 not working

baracoder, 2021-02-20:
Have a similar problem to #5  with BTT SKR Mini 2.0 and TFT35-e3 v3.0

klipper config:
```
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
```
The connector EXT3 on the display is used.

Tested with firmware version 1.2 and readd_cs_pin branch at 3297ceb
I see "ST7920Emulator ready" on the display and with 3927ceb some pixels filling in the top left corner, nothing on the top right

Sometime it will work once after powering on the system or if I disconnect and reconnect the display and use `firmware_restart` on klipper. I don't have any additional devices connected to those pins.

With the touch firmware, the screen stayed black most of the time, but it helped to long-press the knob and switch back and forth between touch and mariln to get the menu back

teeminus, 2021-02-20:
There is a long-press-feature in that branch as well. Press the knob at least 5 seconds. When you release is the screen will turn off and on again. Afterwards please issue a `firmware_restart `. Read here and the following post for more infos: https://github.com/teeminus/NoTouchScreenFirmware/issues/15#issuecomment-781597330

baracoder, 2021-02-21:
Oh, I missed that part. It even starts working even without `firmware_restart` after a few seconds, thanks :tada:

closed 2021-02-21 by baracoder

teeminus, 2021-03-02:
There has been an update to both Klipper and this firmware that should fix your issue. Please checkout the latest release and read the release notes carefully 😄

baracoder, 2021-03-02:
Thank you for the update.
Interestingly it started working better already after a klipper update :thinking: 1.3 also works. I did not update the pin config yet.