teeminus/NoTouchScreenFirmware #21: [request] pin configuration for skr mini 1.2 and tft24

INeDiA, 2021-02-21:
Anyone be able to guess the correct pins for this combination?
The wiring is this one: https://prnt.sc/r2w9yr

Thanks!

teeminus, 2021-02-22:
This looks like you need a special display cable to connect your tft24 to the mainboard. Do you have such a cable?

INeDiA, 2021-02-22:
Of course I do (well, I made it). 
The screen works while using Marlin, both in touch and in emulation mode. But now I switched to Klipper and I can't figure out how to configure pins in my printer.cfg after having flashed your firmware

teeminus, 2021-02-22:
Which display type have you defined in marlin?

teeminus, 2021-02-22:
Are you sure that image is correct? Please check the EXP2 header, the striked out pins do not match with your drawing below.

INeDiA, 2021-02-22:
> Are you sure that image is correct? Please check the EXP2 header, the striked out pins do not match with your drawing below.

I honestly didn't noticed, I just followed the paths to do the wiring!
Is this blueprint helpful? https://github.com/bigtreetech/BIGTREETECH-TFT24-V1.1/blob/master/Hardware/TFT24-V1.1-PIN.pdf

In Marlin I had:
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define CR10_STOCKDISPLAY

Everything was fine, clickwheel included.

Thanks!

teeminus, 2021-02-22:
> Is this blueprint helpful? https://github.com/bigtreetech/BIGTREETECH-TFT24-V1.1/blob/master/Hardware/TFT24-V1.1-PIN.pdf

Yeah, I found that already.

According to you drawing (the lines, not the connectors), this should work
```
[display]
lcd_type: st7920 
cs_pin: PB8
sclk_pin: PB9
sid_pin: PB7
encoder_pins: ^PA9, ^PA10
click_pin: ^!PB6

[output_pin beeper]
pin: PB5
```

INeDiA, 2021-02-22:
Uhmmm I'm afraid this is not working; this is my config:

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PB6, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB7, EXP1_10=<5V>

The display shows "NoTouchFw V1.2" on top and then a big "ST7920Emulator ready" on the screen.
Saved and restarted firmware, also powercycled the screen.

Thanks again <3

teeminus, 2021-02-22:
Are you capable of compiling this firmware yourself? Then checkout the [readd_cs_pin](https://github.com/teeminus/NoTouchScreenFirmware/tree/readd_cs_pin) branch. This could help debug the settings. This branch has an activity indictor for the SPI interface as well as a SPI restart feature when pressing the knob for at least 5 seconds. Refere to [here](https://github.com/teeminus/NoTouchScreenFirmware/issues/15#issuecomment-781597330) for more infos.

INeDiA, 2021-02-22:
The readd_cs_pin is working!
It didn't worked first try, I had to do the 5-seconds-reset and the interface showed up.
It is fully working, temperatures and fans update correctly.

Thanks a lot, feel free to write me if you need any testing with a new release!

[image]

teeminus, 2021-02-22:
> The readd_cs_pin branch is working!

Glad to hear that 😊 In the next release, the SPI indicator will be disabled by default. If you like it (like I do) you have to reenable it yourself.

> It didn't worked first try, I had to do the 5-seconds-reset and the interface showed up.

I am working on a fix for that which will hopefully be included in klipper soon 😉 

> Thanks a lot, feel free to write me if you need any testing with a new release!

I sure will, thanks for the offer 😄

closed 2021-02-22 by teeminus

teeminus, 2021-03-02:
There has been an update to both Klipper and this firmware that should fix your issue. Please checkout the latest release and read the release notes carefully 😄

INeDiA, 2021-03-02:
I flashed the precompiled binary successfully.

"ST7920Emulator ready" (v1.3) is showing at startup, then I always have to reset manually (pressing the knob) at every boot, then turn the knob a few times in order to make the display work.
I think this is working as intended!

teeminus, 2021-03-02:
Have you tried using the `emulated_st7920` lcd_type as described [here](https://github.com/teeminus/NoTouchScreenFirmware/wiki/Klipper-display-configuration#migrate-to-new-display-driver)?

INeDiA, 2021-03-02:
No because I apparently don't pay attention even when people ask me to do so 🤦 

Working config for skr mini 1.2 and tft24, Klipper version: v0.9.1-290 and NoTouchScreenFirmware 1.3:

```
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PB6, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB7, EXP1_10=<5V>
    
[display]
lcd_type: emulated_st7920 
en_pin: PB8
spi_software_sclk_pin: PB9
spi_software_mosi_pin: PB7
encoder_pins: ^PA9, ^PA10
click_pin: ^!PB6
spi_software_miso_pin: PA2 #this pin was unused on my board, change accordingly    
```

teeminus, 2021-03-02:
Thanks for the feedback. Happy printing 😄

INeDiA, 2021-11-03:
sorry for the gravedigging; do you mind adding the config to the wiki? :)

teeminus, 2021-11-03:
Sorry, I totally forgot about that :( Added it just now.