teeminus/NoTouchScreenFirmware #23: Help needed

jappyjan, 2021-02-23:
So im poking around at the TFT35 E3 V3.0 screen for a few hours now and im stuck...

What i got so far:

- I installed the matching noTouchScreenFirmware in Version 1.2 successfully
- i see on screen "ST7920Emulator ready"

But im not getting any further...

Im using a BTT SKR mini e3 v2 with klipper and my klipper config for the screen looks like this:

```
[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
```

teeminus, 2021-02-23:
This is a well known SPI activation problem. There will be an update to both klipper and this firmware in couple of days that addresses this problem. 

The quick and dirty fix (not the final one :D) would be to checkout the [readd_cs_pin](https://github.com/teeminus/NoTouchScreenFirmware/tree/readd_cs_pin) branch and compile the firmware yourself. Afterwards you can press the know for 5 second which will restart the SPI bus and hopefully make the screen work again. For more infos read here: https://github.com/teeminus/NoTouchScreenFirmware/issues/15#issuecomment-781597330

The final fix for this won't require manually restarting the SPI.

teeminus, 2021-02-23:
You can also checkout this issue #20  which seems to be closely related.

jappyjan, 2021-02-23:
so i did the mentioned steps:

- checked out branch "readd_cs_pin"
- build firmware for TFT35_E3_V3_0
- installed/flashed firmware
- restarted spi by holding button for 5 secs

what happened is:
- the screen actually gets blank for a moment and then comes up again, now no longer showing the "Emulator Ready" text but still empty. only thing on screen is the top label "NoTouchFw 1.2" with a line

[image]

i actually tried exp1 and exp3 header on the screen just to make sure... but same result for both.

teeminus, 2021-02-23:
Have you tried executing a `firmware_restart` after the knob long press?

jappyjan, 2021-02-23:
yes

jappyjan, 2021-02-23:
small update.

i again turned off everything and powered everything again.
then i pressed the knob 5 secs to restart the screen.
this time it showed the number "1" and after a few seconds the number "2"

because nothing else happened i run "firmware_reset" again which caused the screen to show again the emulator ready text.
then i pressed the knob again and now its blank again...

jappyjan, 2021-02-23:
ok wow... dunno what changed but did the same once again and now its showing stuff! :D thanks for the help!
i will never turn the printer off again :D

teeminus, 2021-02-23:
Yeah, that bug is pretty annoying. Stay tuned for the final fix which will be available soon 😃

closed 2021-02-23 by teeminus

teeminus, 2021-03-02:
There has been an update to both Klipper and this firmware that should fix your issue. Please checkout the latest release and read the release notes carefully 😄

jappyjan, 2021-03-02:
thank you so much for the effort!
just installed the updates, migrated my config and now everything works flawlessly!

teeminus, 2021-03-02:
Glad to hear that. Thanks for waiting and testing the new release. Have fun 😄