teeminus/NoTouchScreenFirmware #25: Choosing the MISO Pin

Sineos, 2021-03-02:
First off: v1.3 works nicely on my SKR1.4T with E3 V3 and Klipper 290. Many thanks 👍 

Just for clarification and maybe some additional information for the Wiki:

Initially I had this config:
```
[display]
lcd_type: emulated_st7920
spi_software_miso_pin: EXP2_1
spi_software_mosi_pin: EXP1_3
spi_software_sclk_pin: EXP1_5
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
```

`spi_software_miso_pin: EXP2_1`, where `EXP2_1` is actually a MISO pin on the EXP header according to the boards schematic.
It works but I thought it could be causing unwanted side-effects when "overwriting" an actual TFT pin with the software SPI implementation.

[image]

So now I changed to:

```
[display]
lcd_type: emulated_st7920
spi_software_miso_pin: P1.24 #NeoPixel Pin
spi_software_mosi_pin: EXP1_3
spi_software_sclk_pin: EXP1_5
en_pin: EXP1_4
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
```

`P1.24` is on the NeoPixel header, which I do not use. It equally works but might be the "cleaner" solution.

teeminus, 2021-03-02:
Thanks for the suggestion. You could use `EXP1_6` which is the `D4` pin for HD44780 (LCD2004) displays. This pin is dedicated to be used by a display and should therefore be free to use. That's the pin I chose for my miso pin. Could you please try if that works for you as well? If it does I will add both your config and a note on how to choose the miso pin to the wiki.

Sineos, 2021-03-02:
EXP1_6 works equally fine.

Side question:
Actually I did not really try since I use Klipper' virtual SD feature, but does the SD reader on the TFT work in this environment?

teeminus, 2021-03-02:
That's what I asked myself. As far as I understood, there is no support for that.

closed 2021-03-02 by teeminus

teeminus, 2021-03-02:
Btw, I updated the Wiki. And closed this issue 😉 

Thanks for beeing a reliable and motivated tester 😄