teeminus/NoTouchScreenFirmware #30: SKR Mini E3 v2.0 & TFT35 E3 v3.0 - unreliable init, garbled screen

kmarty, 2021-04-23:
_(at this moment, this is probably not an "issue", see the second post. But it might be usefull for the others with the same problem)_

Hi,
I'm having issue with highly unreliable init of TFT35 v3.0 connected to SKR Mini E3 V2.0.
My config is
```
[board_pins]
aliases:
;    # EXP1 header
    EXP1_1=PB5,  EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8,  EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
```
and most of the start of printer, it shows one of the following only:
[image][image]

After shuffling several times with reset(on LCD, which also resets the board -> "FIRMWARE_RESTART" is necessary after this), Klippers' "RESTART" and/or "FIRMWARE_RESTART", the result is either the same or it ends in one of the following:
[image][image]
(about the second one - I'm not sure what happened, this was usually my best result since I switched to NoTouchScreenFirmware till today, when I tried flash back original TFT35 FW... and back to NoTouchScreenFirmware)

or the dream:
[image]

And I am not sure what's wrong. Bug? Bad configuration? Or something else?

kmarty, 2021-04-23:
Today I carefully reread the Wiki (configuration) again and tried "emulated_st7920".
Since SKR Mini E3 v2.0 doesn't have any spare pins for "spi_software_miso_pin", the only "free" for this I found are "TX" and "RX" for "TFT/Touch mode" (which is not used in this case).
So my cfg is now this:
```
[board_pins]
aliases:
;    # EXP1 header
    EXP1_1=PB5,  EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8,  EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>,
    # 
    TFT_TX=PA2,  TFT_RX=PA3

[display]
lcd_type: emulated_st7920
en_pin: EXP1_7
spi_software_sclk_pin: EXP1_6
spi_software_mosi_pin: EXP1_8
spi_software_miso_pin: TFT_RX
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
```
_(not sure which one, TFT_TX or TFT_RX, is better for this while "TFT" cable is remains connected between display and board)_

Anyway, this seems to be a solution. It works reliably, so far.

closed 2021-04-23 by kmarty

teeminus, 2021-04-23:
Hi,

thanks for the feedback.

The config seems fine (it's the same one listed in the wiki). For some printer boards the stock st7920 driver does not work correctly, so using the "emulated_st7920" driver is the only way to make the display work reliably (as you already noticed :D). But I still wonder why you had to use the emulated driver, I thought the SKR Mini E3 would work out of the box. This might be interesting for @emtrax-ltd as well.

The first 3 images indicate the same error: the input signal is shifted by at least one bit. This causes the screen to show garbled data. The fourth image seems like the display actually works but only the graphical display data is missing. I would guess this could have been fixed by switching to the menu by pressing the knob and leaving the menu again.

Glad you got your display working again. Happy printing :)

kmarty, 2021-04-23:
Wiki mentioning SKR Mini E3 v1.2 which is slightly different board.