teeminus/NoTouchScreenFirmware #36: BTT GTR and TFT70v3 Not Working.

jbeima, 2021-05-08:
Now I noticed you have another challenge open in regards to the GTR board, so I am wondering if it may be related to that and not the screen. I did notice some of the GTR pinouts had been confused... So I have more information for this one...

From Marlin -> pinouts -> GTR:

/**
 *               -----                                             -----
 *           NC | · · | GND                                    5V | · · | GND
 *        RESET | · · | PB10(SD_DETECT)             (LCD_D7)  PG5 | · · | PG6  (LCD_D6)
 *   (MOSI)PB15 | · · | PH10(BTN_EN2)               (LCD_D5)  PG7 | · · | PG8  (LCD_D4)
 *  (SD_SS)PB12 | · · | PD10(BTN_EN1)               (LCD_RS)  PA8 | · · | PC10 (LCD_EN)
 *    (SCK)PB13 | · · | PB14(MISO)                 (BTN_ENC) PA15 | · · | PC11  (BEEPER)
 *               -----                                             -----
 *               EXP2                                              EXP1
 */

    #define LCD_PINS_RS                     PA8
    #define BTN_EN1                         PD10
    #define BTN_EN2                         PH10
    #define LCD_PINS_ENABLE                 PC10
    #define LCD_PINS_D4                     PG8

The TFT70v3 does work under Marlin with these settings.

From the Klipper GTR Board Config File:

########################################
# EXP1 / EXP2 pins - GTR Board
########################################

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PC11, EXP1_3=PC10, EXP1_5=PG8, EXP1_7=PG6, EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=PA8, EXP1_6=PG7, EXP1_8=PG5, EXP1_10=<5V>,
    # EXP2 header
    EXP2_1=PB14, EXP2_3=PD10, EXP2_5=PH10, EXP2_7=PB10, EXP2_9=<GND>,
    EXP2_2=PB13, EXP2_4=PB12, EXP2_6=PB15, EXP2_8=<RST>, EXP2_10=<NC>

######################################################################
# "RepRapDiscount 128x64 Full Graphic Smart Controller" type displays
######################################################################

# Display Definitions & Mappings
#display_group: _default_16x4
#cs_pin: LCD_PINS_ENABLE / EXP1_3 / PC10
#sclk_pin: LCD_PINS_D4 / EXP1_5 / PG8
#sid_pin: LCD_PINS_RS / EXP1_4 / PA8
#encoder_pins: BTN_EN1, BTN_EN2 / EXP2_3, EXP2_5 / PD10, PH10
#click_pin: BTN_ENC / EXP1_2 / PA15
#spi_software_sclk_pin: [SCK??|LCD_PINS_D4] / EXP2_2 / PB13 (SCK???)
#spi_software_mosi_pin: MOSI / EXP2_6 / PB15
#spi_software_miso_pin: MISO / EXP2_1 / PB14

######################
# Tried & Did Not Work
#######################

[display]
lcd_type: st7920
#display_group: _default_16x4
cs_pin: EXP1_3
sclk_pin: EXP1_5
sid_pin: EXP1_4
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
#kill_pin: ^!EXP2_8

#[display]
#lcd_type: emulated_st7920
#display_group: _default_16x4
#en_pin: EXP1_3
#spi_software_sclk_pin: EXP1_5
#spi_software_mosi_pin: EXP1_4
#spi_software_miso_pin: EXP1_6
#encoder_pins: ^EXP2_3, ^EXP2_5
#click_pin: ^!EXP1_2

#[display]
#lcd_type: emulated_st7920
#display_group: _default_16x4
#en_pin: EXP1_3
#spi_software_sclk_pin: EXP2_2
#spi_software_mosi_pin: EXP2_6
#spi_software_miso_pin: EXP2_1
#encoder_pins: ^EXP2_3, ^EXP2_5
#click_pin: ^!EXP1_2

######################
# Yet To Try
#######################

#[display]
#lcd_type: emulated_st7920
#display_group: _default_16x4
#en_pin: EXP1_3
#spi_software_sclk_pin: EXP1_5
#spi_software_mosi_pin: EXP2_6
#spi_software_miso_pin: EXP2_1
#encoder_pins: ^EXP2_3, ^EXP2_5
#click_pin: ^!EXP1_2

NOTES:

- I have loaded 1.3 of the firmware onto the TFT70v3 and it does say ST7920 Emulator Ready

jbeima, 2021-05-08:
SOLVED!

I had cs/sid swapped.

I can CONFIRM the following works:

BTT GTR - BTT TFT70V3

```
[display]
lcd_type: st7920
cs_pin: EXP1_4
sclk_pin: EXP1_5
sid_pin: EXP1_3
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
#kill_pin: ^!EXP2_8

[output_pin beeper]
pin: EXP1_1

```

I will add a photo shortly.

jbeima, 2021-05-08:
[image]

closed 2021-05-08 by teeminus

jbeima, 2021-05-08:
I would be happy to try the emulated_st7920, but I would need to confirm the settings. I do believe you need one more?

When I tried this it did not work:

```
[display]
lcd_type: emulated_st7920
en_pin: EXP1_3
spi_software_sclk_pin: EXP1_5
spi_software_mosi_pin: EXP1_4
spi_software_miso_pin: EXP1_6
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
```

I am happy to test variants of it if you may have any suggestions.

teeminus, 2021-05-08:
There is an entry in the wiki on how to convert the config to the emulated driver. If the stock display driver works there is no need to use the emulated driver.