teeminus/NoTouchScreenFirmware #37: "Illegal Flash App" after flashing the firmware.. rendered both displays useless

tnap1979, 2021-05-15:
i have a tft35 v3 e3 and a tft70
flashed with both no touch firmwares, and now both displays show "illegal flash app" on the displays, and the displays no longer function

teeminus, 2021-05-15:
Have you tried a different SD card? Did the SD card you used worked with the original BTT firmware?

tnap1979, 2021-05-15:
i have tried 4 different micro sd..
but when i flash 1.2.. no illegal flash app, and all my displays now work in emulated mode...
tried 1.1 and it flashed, but did not operate, so looks like 1.2 is good enough and works for me..
im using klipper on a tronxy x5sa, ender 3 pro, ender 3 pro v2
tft70 on the x5sa, tft35 v3 e3 on the ender 3 pro v2, tft 50 on the ender 3 pro

teeminus, 2021-05-15:
That's pretty strange. This issue is reported for the original BTT firmware (where this firmware ist based on) as well but there is no real indication what causes the problem. Sometimes using a different SD Card, reformatting the SD, sometimes even recompiling the firmware manually seems to solve the problem. But afaik nobody really knows what causes the problem and as the bootloader of the screens are closed source there is no real way to investigate the problem.

If the v1.2 works for you there is no need to use the latest release.

Would you mind to share your Klipper screen configs so that I can add them to the wiki?

closed 2021-05-20 by teeminus