teeminus/NoTouchScreenFirmware #44: BTT TFT35 E3 not fullscreen

extremeshok, 2021-08-02:
Latest release (1.3) with klipper via a btt skr e3 v2.

It would appear the screen is not sclaed to 100% , but more  like 75%. 

Your banner of the NoTouchFW 1.3 is at 100%

Picture of the screen ... sorry for the bad quality, in an enclosure and still printing.
https://imgur.com/a/WUTkxta?s=wa

extremeshok, 2021-08-02:
Klipper config
```
[display]
lcd_type: emulated_st7920
en_pin: EXP1_7
spi_software_sclk_pin: EXP1_6
spi_software_mosi_pin: EXP1_8
spi_software_miso_pin: EXP1_1 #unused
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
```

teeminus, 2021-08-02:
There is a config Option the the src/User/features.h to enable the fullscreen feature. If you want to use the feature, you can enable the option and compile the firmware yourself.

closed 2021-08-02 by teeminus

extremeshok, 2021-08-02:
Maybe consider setting it as the  default for the next binary release ?

teeminus, 2021-08-02:
Yes, I will definitely do that!