teeminus/NoTouchScreenFirmware #47: Screen will always get garbled after a while.

hapklaar, 2021-08-12:
Used both modes (emulated and not), but in both cases the screen (TFT35-B1 in a Biqu B1) always gets garbled after a little while.

```
#[display]
#lcd_type: st7920 
#cs_pin: EXP1_4
#sclk_pin: EXP1_5
#sid_pin: EXP1_3
#encoder_pins: ^EXP2_5, ^EXP2_3
#click_pin: ^!EXP1_2

[display]
lcd_type: emulated_st7920
spi_software_miso_pin: EXP2_1
spi_software_mosi_pin: EXP1_3
spi_software_sclk_pin: EXP1_5
en_pin: EXP1_4
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2
```

Starting to think it might be some interference going on with the EXP1/EXP2 cables internally, but these issues don't surface when using the normal fw with Marlin mode. This makes me wonder why this implementation suffers from this issue.

teeminus, 2021-08-12:
There seems to be a problem with the Biqu B1 board: https://github.com/teeminus/NoTouchScreenFirmware/discussions/43

Could be a timing issue. The BTT firmware and this firmware share most of the hardware drivers, but there are minor differences in the SPI part. The problem could be caused by this. But other than that, I have no idea what could cause the problem.

hapklaar, 2021-08-12:
Interesting. By the way the same issue occurs when using the 'official' klipper enabled fw from BTT. I also posted this here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1913#issuecomment-897653029

teeminus, 2021-08-12:
Well, thats interesting... 

The improved ST7920 driver (both this firmware and the MR by Msq001 in the original firmware) are more CPU consuming than the previous implementation. Could be a timing issue inside the MCU (e.g. when the I/O resources are used simultaneously, but I am guessing here).

Whats also interesting is, that so far only the Biqu B1 board has been reported to have this issue.

teeminus, 2021-08-12:
Good luck, I hope this will fix the problem :)

BrianValente, 2021-08-16:
I'm having the same problem with the SKR Mini E3 v2 in an Ender 3 Pro

BryanHaven, 2021-08-22:
I was seeing the same issue as @BrianValente - after I did a power cycle it's stuck on the Emulator Ready screen and nothing beyond that - I'm going to try reinstalling the firmware and see if that helps.

hapklaar, 2021-09-11:
> Good luck, I hope this will fix the problem :)

Reporting back; had no effect unfortunately. Is there anything else we can try?

teeminus, 2021-09-11:
When using the emulated_st7920 Klipper display driver, you could try different bus speeds.

hapklaar, 2021-09-11:
How can I do that? Is that an option in klipper's printer.cfg?

teeminus, 2021-09-11:
'spi_speed', see https://www.klipper3d.org/Config_Reference.html?h=spi_speed#common-spi-settings

The default value is 1MHz, minimal value is 100kHz.

teeminus, 2021-09-12:
Maybe the issue is caused by crosstalk on the mainboard.

markcarroll, 2021-09-26:
I am seeing the same problems on a BTT SKR 2 with BTT TFT35 E3 V3.0.  Here are my settings:

```
[display]
lcd_type: emulated_st7920
spi_software_miso_pin: EXP2_1
spi_software_mosi_pin: EXP1_3
spi_software_sclk_pin: EXP1_5
en_pin: EXP1_4
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2
```

It goes wrong for me on homing too, but only when the X and Y axes both move at the same time.  Homing X is fine, homing Y is fine, then it moves the probe to the center to home Z and that is when it garbles.

teeminus, 2021-11-22:
Have tried the original BTT firmware lately? Maybe the original firmware works better 🤷‍♂️

hapklaar, 2021-11-22:
> Have tried the original BTT firmware lately? Maybe the original firmware works better 🤷‍♂️

Actually your current 1.3.1 release works almost flawlessly for me. In about 100 prints I only had one occasion where the screen froze and one where it garbled up, both at the start of a print. This is much better than before, where it went wrong almost 90% of the time.

teeminus, 2021-11-22:
That really sounds like voodoo as there were only minor changes which have nothing to do with the SPI interface...

hapklaar, 2021-11-22:
I know, weird right, have suspected black magic all along... But I'm happy for now ;)

TazerReloaded, 2021-11-24:
Maybe you could try a shielded cable for the display, for example CAT7 LAN. Connect the shield to a earthed case or GND.
Strange anyways, I have stepper and display cables routed together and none of them is shielded, no issues even with 10h+ prints, I even bought longer cables because I wanted my display in front of the printer.

hapklaar, 2021-11-24:
I don't think it's cable related. Have tried many different cable layouts and also ferrite cores etc, that all made no difference at all.

pixeldoc2000, 2021-11-28:
This is really weird. Same issue here with BIQU B1 and TFT35-B1. Maybe it is down to an SPI Bus Problem. Also though about testing ferrite cores and cable routing.

pixeldoc2000, 2021-12-01:
Weird...

Today I dismounted the Display from the Printer and hooked in my Logic Analyzer. The Issue was gone.

Looks like the Problem does not happen when the Display has no connection to the Printer Chassis (metal housing).

Maybe we have some intermittent Ground Loop Issue when the Display has more or less Contact with the Chassis of the Printer, which is connected to the Earth Wire / Neutral Wire (in EU / Germany).

I will have to do some prints to ensure the Problem does not happen this way.

@hapklaar
Maybe you can give it a try...

BryanHaven, 2021-12-01:
Interesting, does anyone who has a replaced the metal bracket with a printed enclosure seem to have this issue?

teeminus, 2021-12-01:
@pixeldoc2000 Very interesting discovery. Could be indeed some kind of ground loop. Or some interferences coupling into the display. Or some difference of potential between the frame and the ground connection of the printer.

My display is electrically isolated from the frame. It's basically dangling on a strain of yarn since my printer enclosure is not done yet 😂

hapklaar, 2021-12-01:
@pixeldoc2000 For unknown reasons I'm no longer experiencing the issue since updating to 1.3.1.

When the issue rears its head again, I will investigate any grounding issues.

PS My B1 is not connected to a grounded mains socket.

pixeldoc2000, 2021-12-01:
@hapklaar
Ok. But why isn't it grounded? In what Country do you live? Sounds unsafe to me...

Did two successful prints in a row and the Display is still not garbled and useable while not connected to the Chassis. I believe it's a first for me. ATM, this seems to fix the issue.

The question remain, why does it happen only with Klipper and not with Marlin? Maybe we will have to dive into the Marlin Source Code to figure out if the SPI Bus or other hardware aspects are handled differently like enable pullups or else.

pixeldoc2000, 2021-12-13:
I've had no issues since the Display is "disconnected" with the Chassis Ground.

teeminus, 2021-12-20:
@hapklaar Did you encounter the issue again or is 1.3.1 still working fine?

hapklaar, 2021-12-20:
> @hapklaar Did you encounter the issue again or is 1.3.1 still working fine?

Haven't had the issue anymore after that one time since I installed 1.3.1.

Revenger, 2022-11-15:
I am getting this myself on my Ender 3 Pro SKR Mini e# 2.0 and TFT35
So wondering if anyone has a solution will have to investigate sometime.

TazerReloaded, 2022-11-15:
> I am getting this myself on my Ender 3 Pro SKR Mini E3 2.0 and TFT35 with Klipper Octoprint on a Pi3B+ So wondering if anyone has a solution will have to investigate sometime.

Is your display PCB connected to the printer frame somewhere (screws)? Mine is held by a printed bracket and has only the wires electrically connected. But I have a TFT24, maybe its hardware related. I also recommend a quality power supply and separate wires for the high power components and the electronics, some chips crash if the power supply is unstable. Maybe a capacitor on the power input side of the display helps?

Revenger, 2022-11-16:
Here's a test I did.
Both screens not connected to the frame, both sitting on a bathroom tile when powered.
BTT TFT 35 is glitchy.
Default screen is fine.

[image]

[image]

teeminus, 2022-11-16:
Thanks for your investigations and your feedback. This supports the suspicion, that it’s a hardware/grounding problem caused by the screen itself rather than the firmware.

Revenger, 2022-11-16:
Went back to Marlin reflashed my Pi to is backup just before Klipper to test a different issue and everything was fine when screen was detected.

[image]

BrockPlaysFortnite, 2023-02-23:
Is there any update on this? I have also had this problem more and more lately. At first it was every now and then but now it's almost every day. 

It happened just now as I opened the printed screen cover for my Ender 3 pro and it made me think it had to be related somehow to the cables or screen itself that slightly wiggled when I opened it since there's no way it happened randomly just as I was opening the screen cover.

teeminus, 2023-02-24:
It's most likely a grounding issue, like a grounding loop that interferes with the data signals. For me, isolating the display from the frame works pretty well.

bkenobi, 2023-03-03:
I just tried my first print with this firmware and it pretty much immediately started glitching.  Is it still believed to be a grounding issue or is there anything else I should consider?  The firmware was sooo much more stable than the BTT firmware in Marlin mode up until trying to print.

bkenobi, 2023-03-04:
I printed a few more parts and this seems random.  Sometimes it works fine.  Other times it will get garbled after it initiates the print (fine during bed leveling).  I'm not sure I can provide anything to help since it seems somewhat random at the moment.

TazerReloaded, 2023-03-04:
Could you check if one of the heaters has an influence on the screen?
- turn heaters on and off, individually or together, and see if the screen has issues
- move the printer around manually, turn lights, fans and other stuff on and off
- if during one of those operations the screen gets garbled, try a "dry print" with that feature disabled. Many slicers can disable extrusion for tests, so you can "print" without heaters.
I sometimes had the garbled screen problem with the old 12864 display and Marlin, which was a hardware implementation, so I don't think it's an issue with the emulation, but with communication or the display driver, most likely electrical (inductive coupling, ground loop, under voltage).
Unfortunately that's very hard to diagnose, and I replaced almost every part of my printer since I had that issue, so I don't know what caused it then and can't test anything myself.
Here is my setup, which has worked without issues for years now, maybe you can test some of those points:
- main board is a BTT SKR 1.4, display is a BTT TFT24
- display is not electrically connected to the aluminium frame
- cables are quite long, around 50cm between main board and display
- cables run alongside other cables (steppers and heaters), but only for a few centimeters, not the entire length
- multi-rail power supply for industrial PCs, separate rails for electronics/steppers and heaters/fans
- configuration is as posted above by markcarroll, stock Klipper, nothing special
- printer frame is not well grounded, many plastic parts between metal components, the heated bed is connected to PE, but everything else is floating

bkenobi, 2023-03-12:
I have not had a chance to perform any of the recommended checks.  However, I did notice my nozzle had backed out a bit requiring a schmoo (PLA) cleanup.  When I touched the hotend with my wrench to remove the nozzle, the display seems to have reset.  Later when I reinstalled the nozzle, the wrench caused it to change a few characters which is what happens during the garbled screen issues.

I had grounded myself to the frame of the Ender3 before touching the wrench to the hotend so I didn't shock anything.  But, I suspect there could be an issue with a lose connection somewhere that causes things to ground.  OR, when I touched the hotend I moved the x-axis which caused some feedback through the motor and made the control board upset.  I'll monitor because the last couple times I've used the printer the display has been working fine.

killrob, 2024-01-09:
i had quite this issue too using an SKR mini E3 V2  and BTT TFT35 E3 V3
this is my config:
`[display]
lcd_type: emulated_st7920
en_pin: EXP1_7
spi_software_sclk_pin: EXP1_6
spi_software_mosi_pin: EXP1_8
spi_software_miso_pin: PA3
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2`

but the screen is black with only "marlin mode" on top screen and nothing else.