teeminus/NoTouchScreenFirmware #59: Option to hide the banner

hapklaar, 2021-11-10:
Hi @teeminus 

The NoTouchFw v1.3.1 banner looks a bit strange on my TFT35B1. (It's compiled in normal mode, I don't like the stretched out Full Screen look.)

- Is it some kind of display corruption, like I'm experiencing more often during printing? (https://github.com/teeminus/NoTouchScreenFirmware/issues/47)
- Is it possible to hide the banner or make an option to do so?

teeminus, 2021-11-10:
The spi activity indicator is working as intended. For each byte received a pixel on the left and right side of the top label is drawn/cleared. As the data is received in blobs one usually does not notice a single pixel of the indicator been drawn/cleared rather than complete lines of the indicator change.

This feature is for debugging purposes only, it is not required for normal operation.

teeminus, 2021-11-10:
It sure is. However, hiding the header makes debugging more complicated. Even when the spi status indicator is not enabled, the header contains valuable information I can use to figure out what might be going wrong.

So adding an option to remove the header would make debugging for those builds more complicated.

It might sound a bit harsh (which is not intentional) but if you really don't like the header, you can easily remove it in your fork.