teeminus/NoTouchScreenFirmware #61: TFT 24 V1.1 not working.

Freudlos, 2022-01-27:
Hey, I flashed the display with the TFT 24 V1.1 bin file and it only bootet up but didnt connect to the Printer. It only displays ST7920Emulator ready.
I have it connected with the BTT Octopus 1.1 and running on klipper.
In the following is my config.
[Octopus current config.txt](https://github.com/teeminus/NoTouchScreenFirmware/files/7949364/Octopus.current.config.txt)

Freudlos, 2022-01-27:
Nvm it worked with:

[display]
lcd_type: emulated_st7920
spi_software_miso_pin: EXP2_1
spi_software_mosi_pin: EXP1_3
spi_software_sclk_pin: EXP1_5
spi_speed: 1000000
en_pin: EXP1_4
encoder_pins: ^EXP2_3, ^EXP2_5
click_pin: ^!EXP1_2

Nevertheless, much respect to you teeminus because you managed to reprogramm the firmware of the TFT screen without it I would have spent 50€ for a useless screen

teeminus, 2022-01-27:
Wow, that's strange. Technically both configs should work.

Anyways, I am glad your screen works now 😃

closed 2022-01-27 by teeminus