teeminus/NoTouchScreenFirmware #63: Knob LED on TFT35 E3 v3

samwiseg0, 2022-05-30:
Everything works great with the firmware. I searched and looked at the code some and could not find on how to control the LED on the knob itself for a TFT35 E3 v3? Is that something the firmware is set to do?

teeminus, 2022-05-30:
The knob is mounted on the TFT PCB but is connected to the 3D printer mainboard directly. The display firmware has nothing to do with the knob.

samwiseg0, 2022-05-30:
> The knob is mounted on the TFT PCB but is connected to the 3D printer mainboard directly. The display firmware has nothing to do with the knob.

Understood. I am referring to the LED itself? Is that controlled via klipper?

teeminus, 2022-05-30:
The original firmware has two modes. The first mode controls the firmware using gcode commands and display the result of the status output. The second mode emulates a ST7920 display (they call it Marlin mode) and needs to be actively controlled by the 3D printer. This firmware supports just the seconds mode.

You need to configure your 3D printer accordingly to use the display in the ST7920 mode. If you are using Klipper, you might find a working config in the wiki.

samwiseg0, 2022-05-30:
> The original firmware has two modes. The first mode controls the firmware using gcode commands and display the result of the status output. The second mode emulates a ST7920 display (they call it [...]

I apologize if I am not making myself clear. I have everything working. I have no issues. There is an LED neopixel that exists for the knob on the TFT35 E3 v3. In the original firmware that was not controlled by marlin but by the firmware itself. What I am asking is can the firmware control this LED? Based off your responses I assume not since its not in the code to control it. If that is the case then I know its minor but I would like to ask if that could be a feature to be considered. 

Info here of what I am referring to with the original firmware. 
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/config.ini#L803-L836

teeminus, 2022-05-30:
Is it just the knob led or the led strip mentioned in the linked file?

If it's just statically turning the knob led on, then I could implement that feature.

samwiseg0, 2022-05-30:
> If it's just statically turning the knob led on, then I could implement that feature.

Yep that is it! Just the LED on the knob itself.

teeminus, 2022-05-30:
With a static color or do you need dimming support ad well?

samwiseg0, 2022-05-30:
> With a static color or do you need dimming support ad well?

Dimming support would be great. The original firmware had it dim with the idle timer. So in this case making it part of `LCD_PWM_DIMMER` would replicate that functionality.

teeminus, 2022-05-30:
Okay, I will see what I can do for you. Give me a few days.

Sorry for being so harsh initially...

samwiseg0, 2022-05-30:
> Okay, I will see what I can do for you. Give me a few days.

No worries! take your time!

> Sorry for being so harsh initially...

Not a problem. I should have more clearly outlined what I was asking.

teeminus, 2022-06-03:
Can you compile the firmware yourself or do you need a precompiled firmware to test?

samwiseg0, 2022-06-06:
> Can you compile the firmware yourself or do you need a precompiled firmware to test?

I can compile it myself.

teeminus, 2022-06-06:
Okay, once I am done I will push my changes to a new branch and notify you for testing.

teeminus, 2022-07-04:
Hi, I just pushed the RBG knob support to a new branch. It needs to be activated in the `features.h` file and is untested (as I don't own a TFT with a RGB knob).

Sorry for letting you wait so long, I was really busy the last time...

AllianceJE, 2022-07-15:
I compiled the RGB branch to test it out but enabling RGB knob results in a blank screen. Disabling RGB knob the firmware works as expected.

teeminus, 2022-07-15:
Yeah, I found a bug which lead to an endless loop. I pushed a fix for that.

AllianceJE, 2022-07-26:
Enabling RGB knob still results in a blank screen

teeminus, 2022-08-07:
Sorry for the delayed replay. I think I fixed the issue, there was a bug in the timing calculation.

AllianceJE, 2023-01-16:
I apologize for taking so long to reply. I’ve been busy with work. Finally had a chance to test again and like to say everything works perfectly now. Thank you for taking the time to add this feature

closed 2023-03-03 by teeminus