teeminus/NoTouchScreenFirmware #74: any chance to get BTT TFT35 V3.01 with GD chip running?

pug304, 2023-04-14:
I've renamed the bin according to BTT GD scheme, TFT ist flashinh but leads to illegal user app. Seconf try with importing the code into vscoce/platformio also no templates for GD chips.

Any possibilty to get the TFT35 V3,01 running?

defekkt, 2024-01-17:
any news on this? not sure how to get this display running with klipper. 
cheers!

teeminus, 2024-01-18:
Hi, this firmware is based on a pretty old version of the original BTT firmware. Therefore, the new displays are not supported. To make this firmware work with new displays one need to rebase this firmware to a more recent version of the BTT firmware. Unfortunatelly, I don't have the time and hardware to do and test those changes.

The basic idea of this firmware is quite simple: Use the BTT system drivers (display, encoder, sd card, ...) with the `src/User/main.cpp` from this project.